strands-project/mongodb_store #123: mongodb_play KeyError

vonovak, 2015-03-16:
Hello, 
I've got a question on the `monngodb_play` script. I recorded some camera data from the `head_xtion_compressed_rgb_compressed` topic. Now when I want to replay it, I run the play script and get the following output.

`Playing back topics set([u'head_xtion_compressed_rgb_compressed'])
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "./mongodb_play.py", line 116, in queue_from_db
    self.publisher = rospy.Publisher(documents[0]["_meta"]["topic"], msg_cls, latch = documents[0]["_meta"]["latch"], queue_size = 10)
KeyError: 'topic'`

Can you tell whether there is something wrong with the script or there is something broken on our side?

Thanks a lot!

marc-hanheide, 2015-11-04:
@vonovak is this still an issue. I have another user reporting the same problem (he'll probably comment on this soon). Did you record this with `mongodb_log`? Did you find a solution for it already?

jeremy-a-asi, 2015-11-04:
I think I have the same problem. It also seems to be recorded on the mongodb_log issue side (https://github.com/timn/ros-mongodb_log/issues/3). It was suggested to me that the general problem is an incompatibility between mongodb_store and mongodb_log packages. My best guess is that mongodb_log should be updated to support the playback better. I hope we can make this happen as I see this as the go to replacement for rosbag. It's much more powerful to have the ability to play the data back and to run queries on the same data.

marc-hanheide, 2015-11-04:
@hawesie can you comment on this. It seems we cannot use `mongodb_store/mongodb_play.py` to play back things that have been recorded with `mongodb_log`, right? Do you agree that we should make the data formats compatible?

hawesie, 2015-11-05:
They definitely *should* be compatible. Either way the playback should be more robust, so I'll look into this.

hawesie, 2015-11-05:
My guess is this is just sloppy engineering on my part. I assume the 'latch' key is always present, but this was only introduced at d2cb06e876ec807b704f6fbb01747d5d967dfd89 and so recordings made before that will cause this error.

hawesie, 2015-11-05:
@jeremy-a-asi @vonovak it would be great if you could test the code at https://github.com/strands-project/mongodb_store/pull/156 to see if this fixes your problem.

jeremy-a-asi, 2015-11-05:
I’ll try that out. Thanks for letting me know.

[cid:image001.png@01D04620.57208070]

Jeremy Adams
SOFTWARE
T

+1.866.881.2171

asirobots.com<http://www.asirobots.com/>

This message contains confidential information and is intended only for the recipient. If you are not the named addressee you should not disseminate, distribute or copy this e-mail. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Finally, the recipient should check this email and any attachments for the presence of viruses. The company accepts no liability for any damage caused by any virus transmitted by this email.

closed 2015-11-06 by marc-hanheide

jeremy-a-asi, 2015-11-06:
One thing I noticed is that this fix is on a hydro branch. I don’t see an indigo branch but there is a jade. Should I able to build and run this branch with ROS indigo release?

[cid:image001.png@01D04620.57208070]

Jeremy Adams
SOFTWARE
T

+1.866.881.2171

asirobots.com<http://www.asirobots.com/>

hawesie, 2015-11-06:
The hydro branch works for indigo too.

jeremy-a-asi, 2015-11-06:
That's what I thought. I tried it with my setup and am still having the same problems with images. I'm using sensor_msgs/CompressedImage message in a topic. 

1. start the database "roslaunch mongodb_store mongodb_store.launch db_path:=/opt/ros/mongodb_store3 db_port:=62345"
2. play an existing bag file to publish the topics
3. start mongodb_log "rosrun mongodb_log mongodb_log.py -a" and let it run for 30 seconds or so
4. stop the bag playback
5. play from the database "rosrun mongodb_store mongodb_play.py" 

After step 5 I don't see topics published containing the sensor_msgs/CompressedImage. I also see the exceptions from #123.

PR #156 (2015-11-06): Added some extra robustness to mongodb_play
Added some extra robustness to mongodb_store. 

Replicator node now should wait for mongo correctly.

mongodb_play does not need latch to be defined in the logged message. This closes #123.

This also now handles skipping empty collections in a cleaner way.