strands-project/mongodb_store #124: PlanningSceneWorld insert and query

sssmanufacturing, 2015-03-31:
Ok this is driving me a bit nuts. Inserting a blank world object then querying it seems to fail. Is it because of the nested octomaps? or something else? If someone could please enlighten me that would be great.

Code to reproduce;

```python
from moveit_msgs.msg import PlanningSceneWorld
from mongodb_store.message_store import MessageStoreProxy
import mongodb_store.util as dc_util

msg_store = MessageStoreProxy()
world = PlanningSceneWorld()   
msg_store.insert(world)

results = msg_store.query(PlanningSceneWorld._type) 
print "world type: %d" % len(results) # Works, has results

world_doc = dc_util.msg_to_document(world)

results = msg_store.query(PlanningSceneWorld._type, world_doc )
print "world: %d" % len(results) # no results

results = msg_store.query(PlanningSceneWorld._type, { "collision_objects" : world_doc["collision_objects"] })
print "world.collision_objects: %d" % len(results) # Works, has results

results = msg_store.query(PlanningSceneWorld._type, { "octomap.octomap" : world_doc["octomap"]["octomap"] })
print "world.octomap.octomap: %d" % len(results) # no results

results = msg_store.query(PlanningSceneWorld._type, { "octomap.octomap.id" : world_doc["octomap"]["octomap"]["id"] })
print "world.octomap.octomap.id: %d" % len(results) # Works, has results

```

sssmanufacturing, 2015-03-31:
Appears to be that the doc dictionary returned from dc_util.msg_to_document does not put things in exactly the same order as what occurs when insert(world) called, which apparently affects the ability for mongodb to call it a match or not.

hawesie, 2015-03-31:
I don't think the order should matter, but I'll take a closer look.

hawesie, 2015-04-02:
This is odd, because if you just create an Octomap, and it and read it out use the doc version, *that* works.

hawesie, 2015-04-02:
Also, this works, so it feels like it is due to the nesting.

```python
msg_store = MessageStoreProxy()
world = PlanningSceneWorld()   

msg_store.insert(world.octomap.octomap)

doc = dc_util.msg_to_document(world.octomap.octomap)

results = msg_store.query(Octomap._type, doc)
print "doc: %d" % len(results) 
```

sssmanufacturing, 2015-04-03:
Hi Nick,

Thanks for taking the time to look at this case. 

I took the output of a working query from dc_util.msg_to_document and rearranged two of the dictionary entries and used that to query the mongodb via msg_store.query and it stopped it from working. This seems to suggest the dictionary order matters? And that the initial insert order of the msg is sometimes different to the query order?

I can post an example of this if required. 

Regards,
Jeremy

hawesie, 2015-04-04:
Yes, please post an example.

hawesie, 2015-04-16:
I should have some time over this weekend to look at this, so if you have a quick example, that would be great.

sssmanufacturing, 2015-04-16:
We have found a solution. The problem is the dictionary order. Using a ordered dictionary solves the problem. I have a forked repo that I will tidy up and submit a pull request soon.

hawesie, 2015-04-16:
Awesome, thanks.