strands-project/mongodb_store #138: request for jade release

k-okada, 2015-07-21:
please release for jade distro 

Cc: @Furushchev

marc-hanheide, 2015-07-23:
We have not tested this on jade AFAIK. Did anyone try? You @k-okada? If I hear that this works out of the box on jade, I'm happy to push for a release.

k-okada, 2015-07-27:
yes, I confirmed that with `catkin run_tests`

hawesie, 2015-08-03:
@marc-hanheide is that enough of a confirmation for you?

hawesie, 2015-08-04:
I've checked locally and things seem ok in Jade. I've created a PR for a Jade release on the main ROS build farm: https://github.com/ros/rosdistro/pull/9065

hawesie, 2015-09-01:
The old release didn't build due to a change in the C++ driver. This new PR should fix that: https://github.com/ros/rosdistro/pull/9350

hawesie, 2015-09-02:
The binaries are now built, awaiting release: http://jenkins.ros.org/job/ros-jade-mongodb-store_binarydeb_vivid_i386/

hawesie, 2015-09-17:
The release is now out!

closed 2015-09-17 by hawesie