strands-project/mongodb_store #145: Replicator node book keeping

hawesie, 2015-10-09:
It would be useful for the replicator node to keep track of what it actually replicates and when. This will allow an admin to check on the the operations performed and take action if necessary. Perhaps we should add a collection to the database where the replication is sent to in order to log replications? What do you think @marc-hanheide @cburbridge @furushchev ?

marc-hanheide, 2015-10-09:
is that not actually what the [oplog](http://docs.mongodb.org/manual/core/replica-set-oplog/) is for?

furushchev, 2015-10-10:
I already do something like this.:
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_robot_common/jsk_robot_startup/lifelog/periodic_replicator_client.py
If we do anyway, It would be better to introduce "dynamic" book keeping. e.g.: start replication only when network traffic is much available, when logging data become big and need to replicate with `delete_after_move=True`, or when just idle.