strands-project/mongodb_store #152: Mongo process error! Exit code=100

qinhaojie0721, 2015-10-28:
It is the first time that the database is used，I create the database path /opt/ros/mongodb_store,then run the instruction rosrun mongodb_store mongodb_server.py

It's the error returned:
[INFO] [WallTime: 1446001409.463378] Mongo server address: bob:62345
[INFO] [WallTime: 1446001409.483994] Found MongoDB version 2.4.9
[INFO] [WallTime: 1446001409.520237] Wed Oct 28 11:03:29.519 [initandlisten] MongoDB starting : pid=7336 port=62345 dbpath=/opt/ros/mongodb_store 64-bit host=jenniful-Lenovo
[INFO] [WallTime: 1446001409.520702] Wed Oct 28 11:03:29.519 [initandlisten] db version v2.4.9
[INFO] [WallTime: 1446001409.520944] Wed Oct 28 11:03:29.519 [initandlisten] git version: nogitversion
[INFO] [WallTime: 1446001409.521211] Wed Oct 28 11:03:29.519 [initandlisten] build info: Linux orlo 3.2.0-58-generic #88-Ubuntu SMP Tue Dec 3 17:37:58 UTC 2013 x86_64 BOOST_LIB_VERSION=1_54
[ERROR] [WallTime: 1446001409.521437] MongoDB process stopped!
[ERROR] [WallTime: 1446001409.521660] Mongo process error! Exit code=100
[INFO] [WallTime: 1446001409.560936] Shutting down datacentre
[WARN] [WallTime: 1446001409.561191] It looks like Mongo already died. Watch out as the DB might need recovery time at next run.

qinhaojie0721, 2015-10-28:
The reason is I don't have the permissions to write to /opt/ros/mongodb_store,run chmod 777 /opt/ros/mongodb_store to solve the problem

closed 2015-10-28 by qinhaojie0721

marc-hanheide, 2015-10-28:
you easily run this as `rosrun mongodb_store mongodb_server.py _database_path:=<your path>` with your own path.

qinhaojie0721, 2015-10-30:
I will be appriciate if you can help me to solve another problem.
After rosrun mongodb_store mongodb_server.py,I execute rosrun mongodb_store message_store_node.py.
I get:

Traceback (most recent call last):
  File "/home/jenniful/catkin_ws/src/mongodb_store/mongodb_store/scripts/message_store_node.py", line 240, in <module>
    store = MessageStore()
  File "/home/jenniful/catkin_ws/src/mongodb_store/mongodb_store/scripts/message_store_node.py", line 41, in __init__
    self._mongo_client=MongoClient(db_host, db_port)
  File "/home/jenniful/catkin_ws/src/mongodb_store/mongodb_store/src/mongodb_store/util.py", line 82, in mongo_client_wrapper
    return pymongo.MongoClient(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pymongo/mongo_client.py", line 352, in __init__
    raise ConnectionFailure(str(e))
pymongo.errors.ConnectionFailure: could not connect to bob:62345: [Errno -2] Name or service not known

I am sure the mongodb_port is 62345 and mongodb_host is bob.

Thanks.

marc-hanheide, 2015-11-02:
To me that seems to be a problem in your network setup. Is there even a machine called "bob" in your setup? And can that name be resolved by the system?

marc-hanheide, 2015-11-02:
So, is "bob" in your /etc/hosts?