strands-project/mongodb_store #157: Test data for mongodb_log and mongodb_play

hawesie, 2015-11-05:
I'd like to create some unit tests for the log and playback nodes. This will require a rosbag of previously logged data. I have a reasonable one of 40MB, but I don't think I should add this to github. @marc-hanheide any thoughts on where I should upload so that it can be easily accessible for jenkins tests *and* for people running tests locally?

marc-hanheide, 2015-11-06:
owncloud? I created a directory [`unit_test_data/`](https://lcas.lincoln.ac.uk/owncloud/index.php/apps/files/?dir=%2FSTRANDS%20data%2Funit_test_data) under `STRANDS data/`. This directory is publicly available at https://lcas.lincoln.ac.uk/owncloud/index.php/s/6hYIdb1Gnk89AUv

Any file you put in there, can directly be downloaded from the URL

`https://lcas.lincoln.ac.uk/owncloud/index.php/s/6hYIdb1Gnk89AUv/download?path=%2F&files=FILENAME`, with `FILENAME` obviously replaced by the actual name of the file, so e.g. https://lcas.lincoln.ac.uk/owncloud/index.php/s/6hYIdb1Gnk89AUv/download?path=%2F&files=dummy.txt will download the file `dummy.txt` I put there for testing. 

In CMake use the `file DOWNLOAD` command to download the file:

```
file(DOWNLOAD url file [INACTIVITY_TIMEOUT timeout]
     [TIMEOUT timeout] [STATUS status] [LOG log] [SHOW_PROGRESS]
     [EXPECTED_HASH ALGO=value] [EXPECTED_MD5 sum]
     [TLS_VERIFY on|off] [TLS_CAINFO file])
```

e.g. 

```
file(DOWNLOAD 
    https://lcas.lincoln.ac.uk/owncloud/index.php/s/6hYIdb1Gnk89AUv/download?path=%2F&files=dummy.txt 
    dummy.txt
)
```

marc-hanheide, 2015-11-06:
this discussion is also probably relevant for @cdondrup who is also implementing tests which rely on some data...