strands-project/mongodb_store #164: i cant run the mongodb_server.

foxnudt, 2016-03-17:
Use "rosrun mongdb_store mongodb_server.py" 
And errors below:
[ERROR] [WallTime: 1458208619.863728] Mongo process error! Exit code=100
 It looks like Mongo already died. Watch out as the DB might need recovery time at next run.

someone help me ?

cdondrup, 2016-03-17:
Sounds like your started it before and didn't finish correctly. Try:

```
ps aux |grep mongod
```

And kill the process that is running. Try to restart the mongodb afterwards.

foxnudt, 2016-03-17:
i've done what you told me ,but ...

[INFO] [WallTime: 1458210254.500315] Mongo server address: localhost:62345
[INFO] [WallTime: 1458210254.526112] Found MongoDB version 2.4.9
[INFO] [WallTime: 1458210254.568952] Thu Mar 17 18:24:14.568 [initandlisten] MongoDB starting : pid=17067 port=62345 dbpath=/opt/ros/mongodb_store 64-bit host=turtlebot
[INFO] [WallTime: 1458210254.569734] Thu Mar 17 18:24:14.568 [initandlisten] db version v2.4.9
[INFO] [WallTime: 1458210254.570227] Thu Mar 17 18:24:14.568 [initandlisten] git version: nogitversion
[INFO] [WallTime: 1458210254.570709] Thu Mar 17 18:24:14.568 [initandlisten] build info: Linux orlo 3.2.0-58-generic #88-Ubuntu SMP Tue Dec 3 17:37:58 UTC 2013 x86_64 BOOST_LIB_VERSION=1_54
[INFO] [WallTime: 1458210254.571129] Thu Mar 17 18:24:14.568 [initandlisten] allocator: tcmalloc
[ERROR] [WallTime: 1458210254.571415] MongoDB process stopped!
[ERROR] [WallTime: 1458210254.571674] Mongo process error! Exit code=100
[INFO] [WallTime: 1458210254.664663] Shutting down datacentre
[WARN] [WallTime: 1458210254.665038] It looks like Mongo already died. Watch out as the DB might need recovery time at next run.

cdondrup, 2016-03-17:
Ok, one hard reset version would be:

Run 
```
ps aux |grep mongo
```

to make sure there really is nothing running, if it is, kill it with `kill -9 <pid>`. After wards remove the `mongod.lock` file from your mongodb directory (the one where the database has been created).

foxnudt, 2016-03-17:
thank you for you patience. i've killed the thread and deleted the mongod.lock, but i still can't solve it.

turtlebot@turtlebot:/opt/ros$ sudo kill -9 17868
turtlebot@turtlebot:/opt/ros$ ps aux | grep mongod
turtleb+ 17874  0.0  0.1  15956  2532 pts/0    S+   18:43   0:00 grep --color=auto mongod
turtlebot@turtlebot:/opt/ros$ cd /var/lib/
turtlebot@turtlebot:/var/lib$ cd mongodb/
turtlebot@turtlebot:/var/lib/mongodb$ ls
journal  local.0  local.ns
turtlebot@turtlebot:/var/lib/mongodb$ rosrun mongodb_store mongodb_server.py
[INFO] [WallTime: 1458211420.455198] Mongo server address: localhost:62345
[INFO] [WallTime: 1458211420.482667] Found MongoDB version 2.4.9
[INFO] [WallTime: 1458211420.525495] Thu Mar 17 18:43:40.524 [initandlisten] MongoDB starting : pid=17910 port=62345 dbpath=/opt/ros/mongodb_store 64-bit host=turtlebot
[INFO] [WallTime: 1458211420.526406] Thu Mar 17 18:43:40.525 [initandlisten] db version v2.4.9
[INFO] [WallTime: 1458211420.526957] Thu Mar 17 18:43:40.525 [initandlisten] git version: nogitversion
[INFO] [WallTime: 1458211420.527497] Thu Mar 17 18:43:40.525 [initandlisten] build info: Linux orlo 3.2.0-58-generic #88-Ubuntu SMP Tue Dec 3 17:37:58 UTC 2013 x86_64 BOOST_LIB_VERSION=1_54
[ERROR] [WallTime: 1458211420.527836] MongoDB process stopped!
[ERROR] [WallTime: 1458211420.528138] Mongo process error! Exit code=100
[INFO] [WallTime: 1458211420.563737] Shutting down datacentre
[WARN] [WallTime: 1458211420.564070] It looks like Mongo already died. Watch out as the DB might need recovery time at next run.

cdondrup, 2016-03-17:
OK, than this is for @hawesie I'm afraid...

hawesie, 2016-03-17:
Let's sanity check things first. What does the following do?

```bash
mkdir /tmp/foo
mongod --dbpath /tmp/foo
```

foxnudt, 2016-03-17:
totally confused......

turtlebot@turtlebot:/tmp$ mongod --dbpath /tmp/foo
Thu Mar 17 18:53:23.464 [initandlisten] MongoDB starting : pid=17921 port=27017 dbpath=/tmp/foo 64-bit host=turtlebot
Thu Mar 17 18:53:23.464 [initandlisten] db version v2.4.9
Thu Mar 17 18:53:23.464 [initandlisten] git version: nogitversion
Thu Mar 17 18:53:23.464 [initandlisten] build info: Linux orlo 3.2.0-58-generic #88-Ubuntu SMP Tue Dec 3 17:37:58 UTC 2013 x86_64 BOOST_LIB_VERSION=1_54
Thu Mar 17 18:53:23.464 [initandlisten] allocator: tcmalloc
Thu Mar 17 18:53:23.464 [initandlisten] options: { dbpath: "/tmp/foo" }
Thu Mar 17 18:53:23.469 [initandlisten] journal dir=/tmp/foo/journal
Thu Mar 17 18:53:23.469 [initandlisten] recover : no journal files present, no recovery needed
Thu Mar 17 18:53:23.948 [FileAllocator] allocating new datafile /tmp/foo/local.ns, filling with zeroes...
Thu Mar 17 18:53:23.948 [FileAllocator] creating directory /tmp/foo/_tmp
Thu Mar 17 18:53:23.954 [FileAllocator] done allocating datafile /tmp/foo/local.ns, size: 16MB,  took 0.002 secs
Thu Mar 17 18:53:23.954 [FileAllocator] allocating new datafile /tmp/foo/local.0, filling with zeroes...
Thu Mar 17 18:53:23.957 [FileAllocator] done allocating datafile /tmp/foo/local.0, size: 64MB,  took 0.002 secs
Thu Mar 17 18:53:23.959 [initandlisten] couldn't unlink socket file /tmp/mongodb-27017.sockerrno:1 Operation not permitted skipping
Thu Mar 17 18:53:23.959 [initandlisten] waiting for connections on port 27017
Thu Mar 17 18:53:23.959 [websvr] admin web console waiting for connections on port 28017
Thu Mar 17 18:53:23.464 [initandlisten] MongoDB starting : pid=17921 port=27017 dbpath=/tmp/foo 64-bit host=turtlebot
Thu Mar 17 18:53:23.464 [initandlisten] db version v2.4.9
Thu Mar 17 18:53:23.464 [initandlisten] git version: nogitversion
Thu Mar 17 18:53:23.464 [initandlisten] build info: Linux orlo 3.2.0-58-generic #88-Ubuntu SMP Tue Dec 3 17:37:58 UTC 2013 x86_64 BOOST_LIB_VERSION=1_54
Thu Mar 17 18:53:23.464 [initandlisten] allocator: tcmalloc
Thu Mar 17 18:53:23.464 [initandlisten] options: { dbpath: "/tmp/foo" }
Thu Mar 17 18:53:23.469 [initandlisten] journal dir=/tmp/foo/journal
Thu Mar 17 18:53:23.469 [initandlisten] recover : no journal files present, no recovery needed
Thu Mar 17 18:53:23.948 [FileAllocator] allocating new datafile /tmp/foo/local.ns, filling with zeroes...
Thu Mar 17 18:53:23.948 [FileAllocator] creating directory /tmp/foo/_tmp
Thu Mar 17 18:53:23.954 [FileAllocator] done allocating datafile /tmp/foo/local.ns, size: 16MB,  took 0.002 secs
Thu Mar 17 18:53:23.954 [FileAllocator] allocating new datafile /tmp/foo/local.0, filling with zeroes...
Thu Mar 17 18:53:23.957 [FileAllocator] done allocating datafile /tmp/foo/local.0, size: 64MB,  took 0.002 secs
Thu Mar 17 18:53:23.959 [initandlisten] couldn't unlink socket file /tmp/mongodb-27017.sockerrno:1 Operation not permitted skipping
Thu Mar 17 18:53:23.959 [initandlisten] waiting for connections on port 27017
Thu Mar 17 18:53:23.959 [websvr] admin web console waiting for connections on port 28017^CThu Mar 17 18:57:03.957 [signalProcessingThread] got signal 2 (Interrupt), will terminate after current cmd ends
Thu Mar 17 18:57:03.957 [signalProcessingThread] now exiting
Thu Mar 17 18:57:03.957 dbexit:
Thu Mar 17 18:57:03.957 [signalProcessingThread] shutdown: going to close listening sockets...
Thu Mar 17 18:57:03.957 [signalProcessingThread] closing listening socket: 9
Thu Mar 17 18:57:03.957 [signalProcessingThread] closing listening socket: 11
Thu Mar 17 18:57:03.957 [signalProcessingThread] shutdown: going to flush diaglog...
Thu Mar 17 18:57:03.957 [signalProcessingThread] shutdown: going to close sockets...
Thu Mar 17 18:57:03.957 [signalProcessingThread] shutdown: waiting for fs preallocator...
Thu Mar 17 18:57:03.957 [signalProcessingThread] shutdown: lock for final commit...
Thu Mar 17 18:57:03.957 [signalProcessingThread] shutdown: final commit...
Thu Mar 17 18:57:03.958 [signalProcessingThread] shutdown: closing all files...
Thu Mar 17 18:57:03.958 [signalProcessingThread] closeAllFiles() finished
Thu Mar 17 18:57:03.958 [signalProcessingThread] journalCleanup...
Thu Mar 17 18:57:03.958 [signalProcessingThread] removeJournalFiles
Thu Mar 17 18:57:03.961 [signalProcessingThread] shutdown: removing fs lock...
Thu Mar 17 18:57:03.961 dbexit: really exiting now

foxnudt, 2016-03-17:
this is my first time to use github, so lucky to have u guys here. but i gotta go now, looking forward to your reply.i'll be online latter @hawesie @cdondrup

hawesie, 2016-03-17:
Did you ctrl-c the process to kill it?

foxnudt, 2016-03-17:
yes

foxnudt, 2016-03-17:
yes

foxnudt, 2016-03-17:
if i didn't press ctrl-c ,it would stay here:
Thu Mar 17 18:53:23.959 [initandlisten] couldn't unlink socket file /tmp/mongodb-27017.sockerrno:1 Operation not permitted skipping
Thu Mar 17 18:53:23.959 [initandlisten] waiting for connections on port 27017
Thu Mar 17 18:53:23.959 [websvr] admin web console waiting for connections on port 28017

LovelyHorse, 2016-04-27:
I had this exact issue and a combination of the following things fixed it:
    #make sure mongod isn't running (I found strange behaviours on my system)
    $ killall -15 mongod

    # first, give yourself write permissions on the folder you've set as the db path. 
    # this may be caused by you creating the folder in /opt/ros/mongodb_store or /tmp/foo with sudo mkdir
    $ sudo chown -R *your username* /data/db

    # If this doesn't work, repair mongod 
    $ rm -rf /*db path*/mongod.lock
    $ mongod --repair

I did the above and it's now working for me. Hopefully that helps!

Vesperal-Hunter, 2016-06-06:
I also have the same problem and I have tried the solutions showed before, but it still not works. If I run the launch file first as the instruction, the errors were showed like these:

`roslaunch mongodb_store mongodb_store.launch db_path:=/home/shuyuhan/test1`
`rosrun mongodb_store mongodb_server.py  `

> [INFO] [WallTime: 1465155227.971993] Mongo server address: shuyu:12345
> [INFO] [WallTime: 1465155227.991608] Found MongoDB version 2.4.9
> [INFO] [WallTime: 1465155228.018190] Sun Jun  5 20:33:48.017 [initandlisten] MongoDB starting : pid=20339 port=12345 dbpath=/opt/ros/mongodb_store 64-bit host=buzz
> [INFO] [WallTime: 1465155228.018580] Sun Jun  5 20:33:48.017 [initandlisten] db version v2.4.9
> [INFO] [WallTime: 1465155228.018771] Sun Jun  5 20:33:48.017 [initandlisten] git version: nogitversion
> [INFO] [WallTime: 1465155228.018979] Sun Jun  5 20:33:48.017 [initandlisten] build info: Linux orlo 3.2.0-58-generic #88-Ubuntu SMP Tue Dec 3 17:37:58 UTC 2013 x86_64 BOOST_LIB_VERSION=1_54
> [INFO] [WallTime: 1465155228.019162] Sun Jun  5 20:33:48.017 [initandlisten] allocator: tcmalloc
> [ERROR] [WallTime: 1465155228.019332] MongoDB process stopped!
> [ERROR] [WallTime: 1465155228.019578] Mongo process error! Exit code=100
> [INFO] [WallTime: 1465155228.020220] Shutting down datacentre
> [WARN] [WallTime: 1465155228.020464] It looks like Mongo already died. Watch out as the DB might need recovery time at next run.

If I run the mongodb_server.py fist, it showed that:
`shuyuhan@buzz:~/catkin_ws/devel/share$ rosrun mongodb_store mongodb_server.py`

> Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.

Also, these two commands are likely to be contrast with each other. Besides, this was what it showed when I check the issues Nick mentioned.

`shuyuhan@buzz:~$ mongod --dbpath /tmp/foo`

> Mon Jun  6 20:55:21.257 [initandlisten] MongoDB starting : pid=31621 port=27017 dbpath=/tmp/foo 64-bit host=buzz
> Mon Jun  6 20:55:21.257 [initandlisten] db version v2.4.9
> Mon Jun  6 20:55:21.257 [initandlisten] git version: nogitversion
> Mon Jun  6 20:55:21.257 [initandlisten] build info: Linux orlo 3.2.0-58-generic #88-Ubuntu SMP Tue Dec 3 17:37:58 UTC 2013 x86_64 BOOST_LIB_VERSION=1_54
> Mon Jun  6 20:55:21.257 [initandlisten] allocator: tcmalloc
> Mon Jun  6 20:55:21.257 [initandlisten] options: { dbpath: "/tmp/foo" }
> Mon Jun  6 20:55:21.291 [initandlisten] journal dir=/tmp/foo/journal
> Mon Jun  6 20:55:21.291 [initandlisten] recover : no journal files present, no recovery needed
> Mon Jun  6 20:55:21.344 [initandlisten] waiting for connections on port 27017
> Mon Jun  6 20:55:21.344 [websvr] admin web console waiting for connections on port 28017

Thanks in advance.

nageshpindi, 2018-09-12:
Can any help me i'm getting error while executing PDDL files

---------------------------------------------------------------------------------------------
process[rosplan_problem_interface-3]: started with pid [12093]
[ INFO] [1536729662.574912724]: KCL: (/rosplan_problem_interface) Ready to receive
process[mongo_server-4]: started with pid [12113]
process[config_manager-5]: started with pid [12120]
process[message_store-6]: started with pid [12121]
process[replicator_node-7]: started with pid [12122]
[rosplan_knowledge_base-2] process has died [pid 12086, exit code -11, cmd /home/rosplan/ROSPlan/devel/lib/rosplan_knowledge_base/knowledgeBase __name:=rosplan_knowledge_base __log:=/home/rosplan/.ros/log/a390df28-b64b-11e8-a24e-080027a2fc5d/rosplan_knowledge_base-2.log].
log file: /home/rosplan/.ros/log/a390df28-b64b-11e8-a24e-080027a2fc5d/rosplan_knowledge_base-2*.log
[ERROR] [1536729663.707063]: 2018-09-12T06:21:03.706+0100 [initandlisten] ERROR: listen(): bind() failed errno:98 Address already in use for socket: 0.0.0.0:62345
[ERROR] [1536729663.707683]: 2018-09-12T06:21:03.707+0100 [initandlisten] ERROR:   addr already in use
[ERROR] [1536729663.752985]: MongoDB process stopped!
[ERROR] [1536729663.753224]: Mongo process error! Exit code=48
[WARN] [1536729663.753635]: It looks like Mongo already died. Watch out as the DB might need recovery time at next run.
[mongo_server-4] process has finished cleanly
log file: /home/rosplan/.ros/log/a390df28-b64b-11e8-a24e-080027a2fc5d/mongo_server-4*.log
[ERROR] [1536729673.641533]: Can't connect to MongoDB server. Make sure mongodb_store/mongodb_server.py node is started.
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/mongodb_store/message_store_node.py", line 305, in <module>
    store = MessageStore()
  File "/opt/ros/kinetic/lib/mongodb_store/message_store_node.py", line 47, in __init__
    raise Exception("No Datacentre?")
Exception: No Datacentre?
[message_store-6] process has died [pid 12121, exit code 1, cmd /opt/ros/kinetic/lib/mongodb_store/message_store_node.py __name:=message_store __log:=/home/rosplan/.ros/log/a390df28-b64b-11e8-a24e-080027a2fc5d/message_store-6.log].
log file: /home/rosplan/.ros/log/a390df28-b64b-11e8-a24e-080027a2fc5d/message_store-6*.log
[ERROR] [1536729723.675774]: Can't connect to MongoDB server. Make sure mongodb_store/mongodb_server.py node is started.
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/mongodb_store/replicator_node.py", line 195, in <module>
    store = Replicator()
  File "/opt/ros/kinetic/lib/mongodb_store/replicator_node.py", line 37, in __init__
    raise Exception("No Datacentre?")
Exception: No Datacentre?

Thanks in advance

hawesie, 2018-09-12:
This line:

> [ERROR] [1536729663.707063]: 2018-09-12T06:21:03.706+0100 [initandlisten] ERROR: listen(): bind() failed errno:98 Address already in use for socket: 0.0.0.0:62345

Makes it look like you already have a mongodb instance running on that port, therefore the node you're launching can't listen there too. Make sure you kill off any other instances. Or if you want to connect to an existing instance use the daemon option.

Aak02ash, 2023-11-23:
Hello,

I'm trying to use mongodb in ROS Noetic but while trying to run the server I'm getting error. The error is similar to the ones mentioned above. i followed the solutions given but it didn't solve my problem. I've cloned the git repo from strands_project.

The error is:

aakash@aakash:~$ rosrun mongodb_store mongodb_server.py
[INFO] [1700757175.067512]: Mongo server address: localhost:62345
[INFO] [1700757175.108404]: Found MongoDB version 3.6.8
[INFO] [1700757175.165452]: 2023-11-23T17:32:55.164+0100 I CONTROL  [initandlisten] MongoDB starting : pid=26445 port=62345 dbpath=/opt/ros/mongodb_store 64-bit host=aakash
[ERROR] [1700757175.167392]: MongoDB process stopped!
[ERROR] [1700757175.168839]: Mongo process error! Exit code=100
[INFO] [1700757175.170756]: Shutting down datacentre
[WARN] [1700757175.172415]: It looks like Mongo already died. Watch out as the DB might need recovery time at next run.

Can someone provide a solution for this? Thanks in advance.