strands-project/mongodb_store #169: Exit code=14 - ARM support?

ButlerRobotics, 2016-05-21:
Does mongodb_store work on ARM?

I have managed to get everything to work on my workstation but when I follow the exact steps on the Jetson TX1 (64-bit ARM) I get an error when trying to run the server:

```
[ERROR] MongoDB process stopped!
[ERROR] Mongo process error! Exit code=14
```

I have seen the other issues with exit code=100 but those solutions do not seem to apply to this case.

Possible issue with MongoDB itself?

hawesie, 2016-05-21:
Please try running `mongod` in its own, and then you’ll get the full error log so we can see if it’s down to mongo or the wrapping we do.

ButlerRobotics, 2016-05-21:
Ah yes, when I run mongod I get a an error:
`Got signal: 7 (Bus error)`

So it must be an issue with MongoDB running on ARM. Looks like I'll have to go back to old SQL :(
Pity, I had everything set up nicely with mongodb_store.

ButlerRobotics, 2016-05-21:
Full output for reference:

```
mongod --help for help and startup options
Sat May 21 10:18:18.618 
Sat May 21 10:18:18.618 warning: 32-bit servers don't have journaling enabled by default. Please use --journal if you want durability.
Sat May 21 10:18:18.618 
Sat May 21 10:18:18.640 Invalid access at address: 0x28e73cb from thread: 

Sat May 21 10:18:18.641 Got signal: 7 (Bus error).

Sat May 21 10:18:18.643 Backtrace:
0x3ddb2c 0x15d374 0x15d522 0xf6f46b10 0x286e44 0x2886c8 0x3dde56 0x14b126 0xf6f37632 
 mongod(_ZN5mongo15printStackTraceERSo+0x17) [0x3ddb2c]
 mongod(_ZN5mongo10abruptQuitEi+0x97) [0x15d374]
 mongod(_ZN5mongo24abruptQuitWithAddrSignalEiP9siginfo_tPv+0x13d) [0x15d522]
 /lib/arm-linux-gnueabihf/libc.so.6(+0x26b10) [0xf6f46b10]
 mongod(_ZN5mongo12BsonUnitTest10testboundsEv+0x7f) [0x286e44]
 mongod(_ZN5mongo12BsonUnitTest3runEv+0x45b) [0x2886c8]
 mongod(_ZN5mongo11StartupTest8runTestsEv+0x25) [0x3dde56]
 mongod(main+0x1ad) [0x14b126]
 /lib/arm-linux-gnueabihf/libc.so.6(__libc_start_main+0x99) [0xf6f37632]
```

hawesie, 2016-05-21:
A bit of googling shows that there are mongodb distributions for ARM, so don’t give up so easily! What OS are you using?

ButlerRobotics, 2016-05-21:
Ah, there is still hope then :)

The TX1 is running on Ubuntu 14.04 Linux 64-bit. But it's been made for the Tegra boards - L4T (Linux 4 Tegra). More details here: [https://developer.nvidia.com/embedded/develop/software]

ButlerRobotics, 2016-05-21:
Any recommendations about which one to try. I am very new to mongodb so am not familiar with building from source and getting mongdb_store to work with it.

hawesie, 2016-05-22:
Try starting from https://github.com/mongodb/mongo/blob/master/docs/building.md

ButlerRobotics, 2016-05-22:
Great thanks, good starting point.

closed 2016-05-22 by ButlerRobotics