strands-project/mongodb_store #171: catkin_make fails on Fedora 23 with ROS Indigo

karpase, 2016-06-22:
Hi,
I tried to build from source on a Fedora 23 machine with ROS Indigo, and got a compilation error.
The first error is:

[ 83%] Building CXX object mongodb_store/mongodb_store/CMakeFiles/message_store.dir/src/message_store.cpp.o
In file included from /usr/include/c++/5.3.1/atomic:38:0,
                 from /usr/include/mongo/platform/atomic_word_cxx11.h:29,
                 from /usr/include/mongo/platform/atomic_word.h:21,
                 from /usr/include/mongo/base/status.h:23,
                 from /usr/include/mongo/util/assert_util.h:23,
                 from /usr/include/mongo/bson/bsontypes.h:20,
                 from /usr/include/mongo/bson/bsonelement.h:25,
                 from /usr/include/mongo/bson/bsonobj.h:27,
                 from /usr/include/mongo/client/bulk_operation_builder.h:20,
                 from /usr/include/mongo/client/dbclientinterface.h:29,
                 from /usr/include/mongo/client/dbclient_rs.h:23,
                 from /usr/include/mongo/client/dbclient.h:59,
                 from /home/robosim/catkin_ws/src/mongodb_store/mongodb_store/include/mongodb_store/message_store.h:14,
                 from /home/robosim/catkin_ws/src/mongodb_store/mongodb_store/src/message_store.cpp:1:
/usr/include/c++/5.3.1/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.

The full log is attached.

karpase, 2016-06-22:
[ros_mongodb_build.txt](https://github.com/strands-project/mongodb_store/files/327698/ros_mongodb_build.txt)
full log is attached here :-(

Thanks,
Erez Karpas

timn, 2016-06-22:
I guess that the issue is that F23 uses the mongo-cxx-client library. This is compiled with the "-std=c++11" flag by default. It is necessary that code using this library is built with the same flag for compatibility...

marc-hanheide, 2016-06-22:
Can you elaborate what you think would have to be changed? Would we have to build mongodb_store with `-std=c++11` as well? I recall this having some issues with ROS and - more definitely - PCL (at least in Ubuntu). Maybe, @karpase can try and just add the flag in your catkin_make call? I think we ought to not add this flag in general, but what does @hawesie think?

karpase, 2016-06-23:
Hi All,
Adding
set(CMAKE_CXX_FLAGS "-std=gnu++11 ${CMAKE_CXX_FLAGS}")

to the top-level CMakeLists.txt helped, and everything now compiles just fine.

You might consider enabling c++11 only when it is supported - see
[here](http://answers.ros.org/question/152276/is-there-a-way-to-enable-c11-support-for-catkin-packages/).

Thanks,
Erez.

marc-hanheide, 2016-06-23:
Great, @timn, thanks for the hint.

closed 2016-06-23 by marc-hanheide

hawesie, 2016-06-23:
Thanks folks. @karpase if you've added that linked fix, please could you open a PR?

karpase, 2016-06-23:
@hawesie I didn't actually add the linked fix, since I'm not quite sure it's the best idea.
Specifically, one of the comments on that page says:
the catkin CMake style guide forbids altering CMAKE_CXX_FLAGS, [cf.](http://docs.ros.org/hydro/api/catkin/html/user_guide/standards.html)