strands-project/mongodb_store #172: Cmake error : MongoClient not found

HuangZengPei, 2016-07-19:
When I try to catkin_make the source code in mu catkin_ws , an error happened. It shows like that:
CMake Error at mongodb_store-hydro-devel/mongodb_store/cmake/FindMongoClient.cmake:72 (message):
MongoClient not found.
Call Stack (most recent call first):
mongodb_store-hydro-devel/mongodb_store/CMakeLists.txt:9 (find_package)
Can anyone tell me how to solve that? Thanks very much

marc-hanheide, 2016-08-09:
This may already answer it for you: http://answers.ros.org/question/217282/mongodb_store-in-ros-indigo-require-mongoclient-package-as-specified-in-the-cmakeliststxt-line-9-where-do-i-can-find-this-package-or-how-can-i-install/

marc-hanheide, 2016-08-09:
If it doesn't please re-open.

closed 2016-08-09 by marc-hanheide