strands-project/mongodb_store #182: Error on launch (MongoQuerywithProjectionMsg)

m312z, 2016-11-28:
Hello all,

I'm getting the error below when trying to launch mongoDB in ROS indigo. Any ideas what is causing this?

Traceback (most recent call last):
  File "/opt/ros/indigo/lib/mongodb_store/message_store_node.py", line 21, in <module>
    class MessageStore(object):
  File "/opt/ros/indigo/lib/mongodb_store/message_store_node.py", line 321, in MessageStore
    query_with_projection_messages_ros_srv.type=dc_srv.MongoQuerywithProjectionMsg
AttributeError: 'module' object has no attribute 'MongoQuerywithProjectionMsg'

Michael

hawesie, 2016-11-28:
Are you using source or an installed package?

m312z, 2016-11-28:
I just installed via:
apt-get install ros-indigo-mongodb-store
I'm launching with:
roslaunch mongodb_store mongodb_store.launch

hawesie, 2016-11-28:
Any other versions of mongodb_store on your machine? This was added in the last release. I’m not sure why it would be missing.

m312z, 2016-11-29:
I uninstalled ros-indigo-mongodb-store and cloned from github. That version worked without issue.

I'll check tomorrow if I still have a hydro version still installed somewhere. It's very possible.

hawesie, 2016-11-29:
My worry is that it's pulling in the python srv module from somewhere old. I'm not sure why it would do that for the services but not the scripts though.

m312z, 2016-12-01:
I have no hydro version anywhere. I'm using 14.04 with only ros-indigo installed.

The clone for github clone is working for me fine.