strands-project/mongodb_store #187: release request for indigo

furushchev, 2017-04-09:
Hi,

Now `hydro-devel` branch includes a feature which is not yet released. (e.g. #184 )
Could you please release a new version if you are ok?
CC: @hawesie 

Thank you very much in advance!

furushchev, 2017-06-17:
@hawesie @marc-hanheide kindly ping to maintainers

hawesie, 2017-06-19:
Ok, on it.

hawesie, 2017-06-19:
Indigo update published. LCAS packages are rebuilt and will release soon. ROS packages are still waiting for the release PR to be tested https://github.com/ros/rosdistro/pull/15283

After these releases I will split mongdb_log out of the repo to start preparing for a Xenial/Kinetic update as discussed in #185

furushchev, 2017-06-19:
@hawesie Thank you very much for releasing!! 👍

closed 2017-06-19 by furushchev