strands-project/mongodb_store #196: check_connection_to_mongod doesn't work with more recent versions of pymongo

jstnhuang, 2017-07-14:
If you are using a recent version of pymongo (I am using v3.4), then [check_connection_to_mongod](https://github.com/strands-project/mongodb_store/blob/hydro-devel/mongodb_store/src/mongodb_store/util.py#L16) doesn't work. Specifically, this line is the problem:

```py
from pymongo import Connection
```

I confirmed that this works with pymongo v2.3 but not with v3.4.

cdondrup, 2017-07-14:
From what I can see in the docs http://api.mongodb.com/python/current/changelog.html#mongoclient-changes the mongo client now starts a connection process in the background that connects to the mongodb as soon as it becomes available. Therefore, no checking of the connection is valid anymore and the Connection class has been removed. 

>MongoClient is now the one and only client class for a standalone server, mongos, or replica set. It includes the functionality that had been split into MongoReplicaSetClient: it can connect to a replica set, discover all its members, and monitor the set for stepdowns, elections, and reconfigs. MongoClient now also supports the full ReadPreference API.
> 
> The obsolete classes MasterSlaveConnection, Connection, and ReplicaSetConnection are removed.
> 
> The MongoClient constructor no longer blocks while connecting to the server or servers, and it no longer raises ConnectionFailure if they are unavailable, nor ConfigurationError if the user’s credentials are wrong. Instead, the constructor returns immediately and launches the connection process on background threads. The connect option is added to control whether these threads are started immediately, or when the client is first used.
> 
> Therefore the alive method is removed since it no longer provides meaningful information; even if the client is disconnected, it may discover a server in time to fulfill the next operation.

Not sure if there is a workaround for this.

jstnhuang, 2017-07-14:
Thanks for the response! I actually don't have a need to check the connection. The problem I encountered is that the message_store_node [performs this check](https://github.com/strands-project/mongodb_store/blob/hydro-devel/mongodb_store/scripts/message_store_node.py#L36) if `mongodb_use_daemon` is true. I just commented this out for my own use, but I thought it'd be helpful to point out.

cdondrup, 2017-07-17:
It is helpful for sure! I'll leave it to @hawesie, @marc-hanheide and the others to decide how to deal with it. But it should be included in the changes being made for #27 imho.

closed 2017-10-30 by hawesie