strands-project/mongodb_store #211: Question about recent change on querying with projection

furushchev, 2018-01-24:
It looks the recent change on https://github.com/strands-project/mongodb_store/commit/1bd27aa73eb2bf9ec52ccba61af6e2ecef86037a just deleted the field for projection from `QueryMsg` and copied it to `QueryWithProjection` and querying with projection was already implemented before the change.
Is there any rationale for creating new method especially for querying with projection?
@hkaraoguz

hawesie, 2018-01-24:
Hmm... good question. I'm afraid I'm been a bit overloaded recently and haven't been checking these as thoroughly as I should have been. @hkaraoguz is there a good motivation for the change?

hkaraoguz, 2018-01-24:
As far as I remember this is done to maintain backward compatibility at that time. Discussion is [here.](https://github.com/strands-project/mongodb_store/pull/173/commits/31c1040e323ee10c7c63a192227a547fe1161a47#diff-16c19150ff2c312e6c136cb217f7b957L20) But this change has happened almost 1.5 years ago. It is not very recent.

furushchev, 2018-01-25:
@hkaraoguz @hawesie Thank you for answering. 

> I remember this is done to maintain backward compatibility at that time.

Got it, is this feature still actively used now, or can we mark it as deprecated?

hkaraoguz, 2018-01-25:
@furushchev No problem. It is still active. It should be working.

furushchev, 2018-02-21:
@hkaraoguz Thanks, got it. Sent a pull request to remove duplicated codes to keep codes simple and now it was merged. Now I close this issue.

closed 2018-02-21 by furushchev