strands-project/mongodb_store #216: Citing this library

furushchev, 2018-02-23:
Hi, I'd like to cite this great library in our papers.
I could not find any citation resource for this library.
Is there any good reference for citing it?
(I prefer BibTeX as follows:)

```
@misc{mongodb_store,
  title="{mongodb\_store}",
  note="\url{https://github.com/strands-project/mongodb_store/}"
}
```

marc-hanheide, 2018-02-24:
(https://zenodo.org/badge/latestdoi/15771698)

marc-hanheide, 2018-02-24:
[image]

marc-hanheide, 2018-02-24:
[image]

marc-hanheide, 2018-02-24:
This DOI will take a bit to be propagated through the system but should provide you with a proper DOI and bibtex eventually.

Thanks

marc-hanheide, 2018-02-24:
https://zenodo.org/record/1183912

marc-hanheide, 2018-02-24:
https://zenodo.org/record/1183912/export/hx

hawesie, 2018-02-24:
You could also cite the paper on the project which caused us to build it! That would probably do our careers more good :) 

http://ieeexplore.ieee.org/document/7948740/

marc-hanheide, 2018-02-24:
As usual @hawesie is thinking a bit further than me.

furushchev, 2018-02-25:
@marc-hanheide @hawesie Thank you very much for answer! I didn't know about DOI / zenodo, this is indeed nice to publish libraries.
(BTW, https://zenodo.org/record/1183912/export/hx shows only nickname for me, I must change my profile on Github :) )

closed 2018-02-25 by furushchev