strands-project/mongodb_store #217: build failed

sunnyamatya, 2018-03-11:
mongodb_store/mongodb_store/CMakeFiles/message_store_cpp_test.dir/build.make:126: recipe for target '/home/sunny/catkin_ws/devel/lib/mongodb_store/message_store_cpp_test' failed
make[2]: *** [/home/sunny/catkin_ws/devel/lib/mongodb_store/message_store_cpp_test] Error 1
CMakeFiles/Makefile2:6969: recipe for target 'mongodb_store/mongodb_store/CMakeFiles/message_store_cpp_test.dir/all' failed
make[1]: *** [mongodb_store/mongodb_store/CMakeFiles/message_store_cpp_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

The file is not available in the folder

hawesie, 2018-03-12:
Please provide a lot more relevant information if you'd like us to help you (e.g. code version, platform/OS, ROS version, etc.)

choiMaster1, 2018-05-02:
code version:0.3.8
platform/OS:ubuntu 16.04 LTS
Ros version: kinetic

mongodb_store/mongodb_store/CMakeFiles/message_store_cpp_test.dir/build.make:126: recipe for target '/home/sunny/catkin_ws/devel/lib/mongodb_store/message_store_cpp_test' failed
make[2]: *** [/home/sunny/catkin_ws/devel/lib/mongodb_store/message_store_cpp_test] Error 1
CMakeFiles/Makefile2:6969: recipe for target 'mongodb_store/mongodb_store/CMakeFiles/message_store_cpp_test.dir/all' failed
make[1]: *** [mongodb_store/mongodb_store/CMakeFiles/message_store_cpp_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

marc-hanheide, 2018-05-17:
This still only looks like the end of the log, not showing the actual error, I think. Can you show a bit more and the exact command you invoked?

marc-hanheide, 2018-05-17:
Btw, you don't need to compile from there, you can also just `apt install ros-kinetic-mongodb-store`