strands-project/mongodb_store #221: Release request

furushchev, 2018-05-02:
Hi, since there are some major changes including message definition updates, I think we need to release the new version.

- mongodb_store: support filter by query on moving entries
- mongodb_store: test replication
- Import connection failure
- mongodb_store: remove duplicated codes for querying with projection
- mongodb_store: message_store_node.py: deprecate using extra servers
- mongodb_store: encode utf-8 on filling rosmsg
- fix error on querying extra servers

hawesie, 2018-05-02:
I should have done this earlier. I'll do it soon!

hawesie, 2018-05-02:
Kinetic: https://github.com/ros/rosdistro/pull/17697

Indigo: https://github.com/ros/rosdistro/pull/17698

Sorry for the delay! It should be quicker in the future as I've now got my docker install set up cleanly.

hawesie, 2018-05-02:
All done.

closed 2018-05-02 by hawesie

furushchev, 2018-05-02:
@hawesie Thank you very much for your really quick response! :+1: