strands-project/mongodb_store #222: release with C++11 compatibility?

bbferka, 2018-05-22:
Hi, 

I was wondering if it would be possible to release libmongocxx_ros with the ``c++11=on``? Would this introduce any unforeseen issues? We are planning on releasing our [project](www.github.org/robosherlock/robosherlock.git) as a binary for kinetic but it depends monogcxx and needs C++11. As it currently is we migrated from `26compat` to `legacy`, but we needed to compile it from source with the c++11 flags, otherwise we get undefined behavior, probably because of the [C++11 ABI compatibility breaks](https://gcc.gnu.org/wiki/Cxx11AbiCompatibility)

hawesie, 2018-05-22:
@furushchev @marc-hanheide any thoughts? I have no objections.

furushchev, 2018-05-22:
Since indigo has no c++11 support, we will have to release separately if needed, but for other things, It is ok to me.
@bbferka Could you open a pull request? We have test codes so we can see how it goes if c++11 is enabled.

bbferka, 2018-05-22:
@furushchev @hawesie thanks for looking into this. I'll open a PR with the changes. You are right. This might not work for indigo.

furushchev, 2018-05-23:
Solved by #223

closed 2018-05-23 by furushchev

furushchev, 2018-06-27:
FYI, this issue is also addressed in #224 and #225