strands-project/mongodb_store #227: CMake Error while building

aarushsesto, 2018-06-28:
I'm getting this error while building the repo

```
CMake Error at mongodb_store/libmongocxx_ros/CMakeLists.txt:91 (add_library):
  No SOURCES given to target: mongoclient

-- Build files have been written to: /home/batman/catkin_ws/build
Invoking "cmake" failed
```

I'm building it on kinetic running on Ubuntu 16.04. My cmake version is 3.11.1

It seems some kind of syntax problem, due to different version of cmake or something like that to me, am I right?

hawesie, 2018-06-28:
Do you have cmake installed? You can do this as below, although I thought rosdep should install this for you.

```
sudo apt-get install cmake
```

aarushsesto, 2018-06-28:
yes I do have cmake installed, as I mentioned above I have cmake version 3.11.1
The error is coming on this line https://github.com/strands-project/mongodb_store/blob/1fadc1e90701b48475800828e16b80ba9aa583db/libmongocxx_ros/CMakeLists.txt#L91 

I'm not familiar with this syntax, in which you haven't added a source code for the library but just the library name? Am I doing something wrong?

hawesie, 2018-06-28:
Please can you provide the log of the build?

aarushsesto, 2018-06-28:
ok so, problem solved, I was able to build it with cmake version 3.5.2, I don't think this syntax is supported with cmake version 3.11.1

closed 2018-06-28 by aarushsesto