strands-project/mongodb_store #228: Can't log compressed color images from kinect

countdueki, 2018-06-29:
Hello everyone,

I tried to record images from the kinect using mongodb_log (from source). Most of the topics work just fine, but when I try to log the _compressed_ color image topic using `rosrun mongodb_log mongodb_log.py /camera/rgb/image_raw/compressed` I get:
`Adding topic /camera/rgb/image_raw/compressed
DETECTED compressed image topic /camera/rgb/image_raw/compressed, using fast C++ logger
mongodb_log_cimg: /usr/include/boost/smart_ptr/scoped_ptr.hpp:99: T* boost::scoped_ptr<T>::operator->() const [with T = mongo::AtomicWord<unsigned int>]: Assertion `px != 0' failed.
`
And no entries are created in the mongodb. Same for topic `/camera/depth_registered/image_raw/compressed`

I use **ROS Kinect** and **Ubuntu 16.04**

PS.:
when I run the kinect with the **freenect-launch** package (roslaunch freenect_launch freenect-registered-xyzrgb.launch)
I get those warnings on startup:
`[ WARN] [1530279220.311538845]: Could not find any compatible depth output mode for 1. Falling back to default depth output mode 1.
[ INFO] [1530279220.317296436]: rgb_frame_id = 'camera_rgb_optical_frame'
[ INFO] [1530279220.317347058]: depth_frame_id = 'camera_depth_optical_frame'
[ WARN] [1530279220.325129058]: Camera calibration file /home/tammo/.ros/camera_info/rgb_B00365704739132B.yaml not found.
[ WARN] [1530279220.325179201]: Using default parameters for RGB camera calibration.
[ WARN] [1530279220.325210333]: Camera calibration file /home/tammo/.ros/camera_info/depth_B00365704739132B.yaml not found.
[ WARN] [1530279220.325239629]: Using default parameters for IR camera calibration.
[ INFO] [1530279222.588314297]: Stopping device RGB and Depth stream flush.
`

hawesie, 2018-07-04:
You should be able to run `mongodb_log_cimg` from the command line for just this topic. If you do that, do you see any other errors? Could you run it in a debugger to see where the error is happening in the code?

countdueki, 2018-07-04:
Thanks for the reply!
When I run `mongodb_log_cimg` from command line I get the same error. My debugger gives me this
`Thread 1 "mongodb_log_cim" hit Breakpoint 1, msg_callback (msg=...) at /home/tammo/catkin_ws/src/mongodb_store/mongodb_log/src/mongodb_log_cimg.cpp:56
56	  mongodb_conn->insert(collection, document.obj());
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)
[Switching to thread 7 (Thread 0x7f2e8960d700 (LWP 5492))](running)

Thread 1 "mongodb_log_cim" received signal SIGABRT, Aborted.
0x00007f2e90af3428 in raise () from /lib/x86_64-linux-gnu/libc.so.6`

But I have reason to think that the problem lies somewhere in the bsonobjbuilder.h or bsonobj.h in the libmongocxx package. 

It may be of importance to mention that  I use mongodb Version 2.6.10 as of now.

countdueki, 2018-07-08:
I get a warning while compiling, that the auto pointer is deprecated. I changed them to unique_ptr without any effect. But I think you might be right with the problem lying in this line. The pointer has some negative values, which are not cast correctly. It has some negative values, which probably lead to the error I get.

hawesie, 2018-07-09:
So there's a problem with the input then?

bbferka, 2018-07-10:
hi all; after quickly looking at this and testing it with a RealSense and a the Kinect on the PR2 I was experiencing the same problem;

The issue is that with legacy cxx drivers you need to call `mongo::client::initialize();` before you can start logging; 

Also the collection name is not set and no error message is shown except the exception that mongo throws. 

I fixed this with #229

countdueki, 2018-07-10:
@bbferka that did the trick! I only had to add the database and collection, seperated by a dot. So e.g. in my case I added ` -c db.kinect_collection` 
so the full command was `rosrun mongodb_log mongodb_log_cimg -t /camera/rgb/image_color/compressed -n camera -c db.kinect_log`
It started logging and the collections seemed to be complete. Appreciate the fast responses!

For completeness:  Only using ` -c kinect_collection` gave the error 
`terminate called after throwing an instance of 'mongo::MsgAssertionException'
  what():  nsToCollectionSubstring: no . `

bbferka, 2018-07-16:
@countdueki glad to hear that; @hawesie I will try to figure out a way so this fix can also compile on Indigo, since currently the build fails (see #229 for details )

chris2401, 2019-05-10:
@bbferka could you please explain in more detail how you got it working?
i am running ros kinetic on ubuntu 16.04. with mongodb-store version 0.4.4 and i want to capture compressed color images of a realsense d435.

running:
`rosrun mongodb_log mongodb_log.py -e /camera/color/image_raw/compressed`
results in
`Adding topic /camera/color/image_raw/compressed
DETECTED compressed image topic /camera/color/image_raw/compressed, using fast C++ logger
mongodb_log_cimg: /usr/include/boost/smart_ptr/scoped_ptr.hpp:99: T* boost::scoped_ptr<T>::operator->() const [with T = mongo::AtomicWord<unsigned int>]: Assertion `px != 0' failed.
`

running 
`rosrun mongodb_log mongodb_log_cimg -t /camera/color/image_raw/compressed -n camera -c db.realsense_log`
results in
`mongodb_log_cimg: /usr/include/boost/smart_ptr/scoped_ptr.hpp:99: T* boost::scoped_ptr<T>::operator->() const [with T = mongo::AtomicWord<unsigned int>]: Assertion `px != 0' failed.
Aborted (core dumped)`

I am unsure how to debug further from here.