strands-project/mongodb_store #230: ROS melodic support

bbferka, 2018-07-16:
Hi, 

I've created a branch (https://github.com/bbferka/mongodb_store/tree/melodic-devel)  on my fork for melodic. As of Ubuntu bionic libmongocxx legacy can be a system dependency (Yay!).  So far I have only run it with unit tests, which all run through OK.  

For now I've only added a CATKIN_IGNORE to libmongocxx_ros, but I think it could also be removed in the future.  

If there is an appropriate branch to which I can send a PR I would gladly do so.

hawesie, 2018-07-17:
Created: https://github.com/strands-project/mongodb_store/tree/melodic-devel

bbferka, 2018-07-17:
#231

abhishek098, 2018-11-27:
Tested the branch [https://github.com/bbferka/mongodb_store.git](https://github.com/bbferka/mongodb_store.git ). Build successful. Please merge asap.

hawesie, 2018-11-27:
@abhishek098 I'll take a look when I can and merge.

hawesie, 2018-11-27:
I've merged it into the branch here. Do you also want this released to the ros build farm?

abhishek098, 2018-11-28:
@hawesie yes. ros release would be great.

hawesie, 2018-11-28:
Noted!

hawesie, 2018-12-20:
I'm part way there. Just getting the pymongo rosdep added for bionic: https://github.com/ros/rosdistro/pull/19807

hawesie, 2019-01-24:
PR opened: https://github.com/ros/rosdistro/pull/20048

hawesie, 2019-01-24:
PR merged so this should start making its way into the build farm

closed 2019-01-24 by hawesie