strands-project/mongodb_store #237: Handling removed default parameters

jorgenfb, 2019-01-31:
We have some parameters previously stored in default_path directory. Now we want to remove some of them from the config manager and instead set them in our launch files. The problem is that the config_manager still applies the old defaults on the parameters server. 

How do I handle this properly assuming I don't have access to the database on all installations. The only workaround I can think of is renaming my parameters, but that will still leave the old ones on the parameter server, which may be very confusing.

hawesie, 2019-02-01:
A good question. Adding @Jailander as I think he's the one who bests understands this functionality. (Jaime, add someone else if not!)

That said, if your nodes can read the parameters from the store, you must have database access in some form, so it should be possible to script something to remove these old database entries.

jorgenfb, 2019-02-01:
I do have database access, I just want to have your opinion on how to handle removed defaults. If there is some missing functionality I might be able to provide it in an PR if we can agree on the desired behavior. 

I think that at least the default value stored in the database should be ignored. I'm not sure what to do with any local override as there is possible to set local configs even for parameters without a default. 

Why is defaults stored in the database anyway? Aren't they read from file anyway?

jorgenfb, 2019-10-28:
I would basically read all the defaults from the database and see if any of them is no longer read from default files, then delete the from the database itself.

I'm not really sure how to handle local config. I would probably still allow to have local config without a default. With #248 we already have a service for that can be used to reset the local values.