strands-project/mongodb_store #246: Repository not registered for documentation indexing?

gavanderhoorn, 2019-04-06:
Just noticed that the [wiki page](http://wiki.ros.org/mongodb_store) doesn't show a `Documentation` badage.

[rosdistro/kinetic/distribution.yaml](https://github.com/ros/rosdistro/blob/c12de20486d116fdcd3eca43aefca5f61b2cc42e/kinetic/distribution.yaml#L6681-L6696) also shows no `doc` entry for the package(s).

Is this intentional?

Comparing this to other ROS wiki pages it makes it harder to find the source repository (I'm lazy, so I clicked on "Use GitHub to report bugs or submit feature requests" and then ended up at the issue tracker).

hawesie, 2019-04-06:
We do actually have some documentation, so I'll add it to the next release.

gavanderhoorn, 2019-04-06:
Doc indexing doesn't depend on having documentation available. Even (almost) empty repositories can be indexed by the ROS buildfarm.

The benefit of having the doc jobs generated is that the *Package Summary* heading on the wiki page auto-generates some additional information about the repository.

The only thing that would be needed would be a `doc` entry added for the repository. See [this entry](https://github.com/ros/rosdistro/blob/c12de20486d116fdcd3eca43aefca5f61b2cc42e/kinetic/distribution.yaml#L10-L13) for an example. A simple PR adding this would be sufficient.

hawesie, 2019-04-06:
Useful to know, thanks. I'll add this to me todo list.

hawesie, 2019-10-23:
kinetic and melodic releases are already indexed for documentaiton, but indigo wasn't and that's the base entry on the wiki. Fix added: https://github.com/ros/rosdistro/pull/22736

hawesie, 2019-10-23:
Hmmm.... it does now... odd.