strands-project/mongodb_store #247: installing from source

summer1216, 2019-04-09:
Due to reasons I have to install mongodb_store from source. 

I cloned it to my catkin_ws and built it by catkin_make.

Then I need to specify the path for this package, but I don't know where is the package installed. Any idears? thanks!

hawesie, 2019-04-14:
Please provide some output from your build process and also details of your OS, ROS version etc.

summer1216, 2019-04-16:
I reinstalled ROS and the problem was solved. Thank you anyway! 

This issue can be closed.

bilal9876, 2020-02-27:
i use git clone  https://github.com/strands-project/mongodb_store.git (install from the source)
in my own catkin workspace . when i execute catkin_make (build) 
Switched to a new branch 'releases/legacy'
Branch 'releases/legacy' set up to track remote branch 'releases/legacy' from 'origin'.

i want to use mongodb with rosplan and melodic and turtlebot3.
any help please?

marc-hanheide, 2020-02-29:
@bilal9876  can't you just use the released package (https://wiki.ros.org/mongodb_store) for melodic?