strands-project/mongodb_store #251: Cannot apt install on melodic

Affonso-Gui, 2019-06-17:
Hello,

I was not able to `apt install ros-melodic-mongodb-store`, apparently because the build is failing
http://repositories.ros.org/status_page/ros_melodic_default.html?q=mongodb

It seemed to be fine when installing from source, so maybe some dependency problems?

Affonso-Gui, 2019-06-17:
@knorth55

knorth55, 2019-06-17:
```
08:31:41 CMake Error at cmake/FindMongoClient.cmake:72 (message):
08:31:41   MongoClient not found.
08:31:41 Call Stack (most recent call first):
08:31:41   CMakeLists.txt:27 (find_package)
08:31:41 
08:31:41 
08:31:41 -- Configuring incomplete, errors occurred!
08:31:41 See also "/tmp/binarydeb/ros-melodic-mongodb-store-0.5.0/obj-x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
08:31:41 See also "/tmp/binarydeb/ros-melodic-mongodb-store-0.5.0/obj-x86_64-linux-gnu/CMakeFiles/CMakeError.log".
08:31:41 	cd obj-x86_64-linux-gnu && tail -v -n \+0 CMakeCache.txt
```

This error is solved in #250 .

```
08:31:41 /usr/bin/cc -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-melodic-mongodb-store-0.5.0=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_b1faa.dir/CheckFunctionExists.c.o  -o cmTC_b1faa -lpthreads 
08:31:41 /usr/bin/ld: cannot find -lpthreads
08:31:41 collect2: error: ld returned 1 exit status
08:31:41 CMakeFiles/cmTC_b1faa.dir/build.make:97: recipe for target 'cmTC_b1faa' failed
08:31:41 make[3]: *** [cmTC_b1faa] Error 1
08:31:41 make[3]: Leaving directory '/tmp/binarydeb/ros-melodic-mongodb-store-0.5.0/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
08:31:41 Makefile:126: recipe for target 'cmTC_b1faa/fast' failed
08:31:41 make[2]: *** [cmTC_b1faa/fast] Error 2
08:31:41 make[2]: Leaving directory '/tmp/binarydeb/ros-melodic-mongodb-store-0.5.0/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
```

I have no idea about this error.

hawesie, 2019-06-17:
Looks like I need to rerelease the melodic version with #250

knorth55, 2019-06-21:
any update for this issue?
I'm wondering if these two error caused by the dependency problem.

hawesie, 2019-06-21:
I should be able to do the re-release on MOnday (don't have access to the correct bits at the moment...).

knorth55, 2019-06-21:
Ok, thank you for your quick response. :+1:

hawesie, 2019-06-28:
Slightly later than planned: https://github.com/ros/rosdistro/pull/21641

knorth55, 2019-06-28:
GOOD!! Thank you for your quick release!

knorth55, 2019-08-29:
@Affonso-Gui we can close this issue.

hawesie, 2019-08-29:
👍

closed 2019-08-29 by hawesie

Affonso-Gui, 2019-09-02:
@hawesie @knorth55  Thanks for your help!!