strands-project/mongodb_store #256: mongodb_store build fail

bilal9876, 2020-02-27:
CMake Error at mongodb_store-upstream-0.5.2/mongodb_store/cmake/FindMongoClient.cmake:72 (message):
  MongoClient not found.
Call Stack (most recent call first):
  mongodb_store-upstream-0.5.2/mongodb_store/CMakeLists.txt:27 (find_package)

error when build the new released version for mlodic ros version

closed 2020-02-28 by bilal9876

ysl208, 2020-03-10:
Can I ask what the solution to this is? I have encountered the same problem

bilal9876, 2020-03-10:
for sure , what i rememer is:

i uninstall all the mongodb and all its prerequesisties.

i download the package from  strands-project/mongodb_store

and follow the steps from readme.txt in the mongodb_store folder.

please reply for any later errors