strands-project/mongodb_store #260: C++ JSON serialization of ROS messages

nilsbore, 2020-05-07:
Hi guys! I implemented a [C++ serializer/deserializer for ROS msgs](https://github.com/nilsbore/roscpp_json_serialize) for a [project I'm working on atm](https://github.com/nilsbore/roswasm/blob/master/src/test.cpp). From what I remember, mongodb_store does that part in python and then sends it on to c++ for insertion in mongodb. Would there be any obvious gain from doing it in c++ instead (e.g. speed, less traffic?). In case you would be interested, I think I could get serialization to be compliant with whatever you're using. There is already a [test suite](https://github.com/nilsbore/roscpp_json_serialize/blob/master/src/test.cpp) that makes sure it does not corrupt **most** messages. Cheers!

hawesie, 2020-05-08:
Hey @nilsbore lovely to hear from you!

Both C++ and Python do it natively using ROS code. Here are the C++ bits: https://github.com/strands-project/mongodb_store/blob/melodic-devel/mongodb_store/include/mongodb_store/message_store.h#L33

nilsbore, 2020-05-08:
Hi @hawesie ! Ok, I see! But from what I understand this is binary serialization of the messages. Wasn't there an option to also have them as json dicts within mongodb?

nilsbore, 2020-05-08:
Basically, my library does the same things as the two functions you pointed to, put serializing to json.

hawesie, 2020-05-15:
Understood! 

That would probably be nicer, but I'm not sure it's necessary at this point (everything works, no-one is actively changing this code). When the message infrastrucutre changes for ROS2, it would be worth looking again.

closed 2020-05-15 by hawesie