strands-project/mongodb_store #267: mongodb for ROS noetic

talhakarimtk, 2021-02-16:
I am trying to access mongodb using ros noetic. Then I found [here](https://answers.ros.org/question/364039/how-to-install-mongodb-for-ros-noetic/) that mongodb package for noetic has not been released. Will the package be released for noetic or should I need to install some previous version of ROS?

hawesie, 2021-02-16:
You can install mongodb_store from source by cloning this repo into your catkin workspace. I haven't tested whether it compiles for noetic, so please let us know how it goes.

shu-qing, 2021-02-19:
$ roslaunch mongodb_store mongodb_store.launch db_path:=/home/sq/Desktop/db db_port:=62345
... logging to /home/sq/.ros/log/abb0ea44-72ab-11eb-a629-3f36563baa48/roslaunch-tp-44094.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tp:34025/

SUMMARY
========

PARAMETERS
 * /config_manager/defaults_path: 
 * /message_store/mongodb_use_localdatacenter: True
 * /message_store/queue_size: 100
 * /mongo_server/bind_to_host: False
 * /mongo_server/database_path: /home/sq/Desktop/db
 * /mongodb_connection_string: 
 * /mongodb_host: localhost
 * /mongodb_port: 62345
 * /mongodb_use_daemon: False
 * /replicator_node/replicator_dump_path: /tmp/replicator_d...
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /
    config_manager (mongodb_store/config_manager.py)
    message_store (mongodb_store/message_store_node.py)
    mongo_server (mongodb_store/mongodb_server.py)
    replicator_node (mongodb_store/replicator_node.py)

auto-starting new master
process[master]: started with pid [44103]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to abb0ea44-72ab-11eb-a629-3f36563baa48
process[rosout-1]: started with pid [44113]
started core service [/rosout]
process[mongo_server-2]: started with pid [44120]
process[config_manager-3]: started with pid [44121]
process[message_store-4]: started with pid [44122]
process[replicator_node-5]: started with pid [44123]
/home/sq/ros/mongodb/src/mongodb_store/mongodb_store/scripts/config_manager.py:114: DeprecationWarning: add_son_manipulator is deprecated
  self._database.add_son_manipulator(MongoTransformer())

hawesie, 2021-02-24:
Ok, that looks good!

gautz, 2021-03-05:
I cloned the repo in a noetic catkin workspace. Doing a ``rosdep install`` gave me:
``mongodb_store: No definition of [python-future] for OS version [focal]``
Then I did ``sudo apt-get install mongodb mongodb-dev python3-pymongo``
I cloned also https://github.com/ros-planning/warehouse_ros_mongo noetic-devel and https://github.com/ros-planning/warehouse_ros kinetic-devel
And it worked.

BlazPotokar, 2022-06-03:
I tested it out and it dones't fully work. Querying messages seem to work in most cases but it returns generator object instead of list. So result needs to be converted to a list to have the same behavior. Some calls like `MessageStoreProxy().query(..., single=True)` fails because of that.
Inserting messages on the other hand, doesn't seem to work at all. I always get serialization error. I tried to use `BytesIO` isntead of `StringIO` as a buffer in `utills.serialise_message` but it crashed elsewhere.
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/root/.vscode-server/extensions/ms-python.python-2022.6.3/pythonFiles/lib/python/debugpy/__main__.py", line 45, in <module>
    cli.main()
  File "/root/.vscode-server/extensions/ms-python.python-2022.6.3/pythonFiles/lib/python/debugpy/../debugpy/server/cli.py", line 444, in main
    run()
  File "/root/.vscode-server/extensions/ms-python.python-2022.6.3/pythonFiles/lib/python/debugpy/../debugpy/server/cli.py", line 285, in run_file
    runpy.run_path(target_as_str, run_name=compat.force_str("__main__"))
  File "/usr/lib/python3.8/runpy.py", line 265, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "/usr/lib/python3.8/runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/test_ws/src/db_test/scripts/test.py", line 22, in <module>
    ms.insert_named('msg', msg)
  File "/test_ws/src/mongodb_store/mongodb_store/src/mongodb_store/message_store.py", line 88, in insert_named
    return self.insert(message, meta_copy, wait=wait)
  File "/test_ws/src/mongodb_store/mongodb_store/src/mongodb_store/message_store.py", line 106, in insert
    serialised_msg = dc_util.serialise_message(message)
  File "/test_ws/src/mongodb_store/mongodb_store/src/mongodb_store/util.py", line 528, in serialise_message
    message.serialize(buf)
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/_Int64.py", line 55, in serialize
    except TypeError as te: self._check_types(ValueError("%s: '%s' when writing '%s'" % (type(te), str(te), str(locals().get('_x', self)))))
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 395, in _check_types
    raise SerializationError(str(exc))
genpy.message.SerializationError: <class 'TypeError'>: 'string argument expected, got 'bytes'' when writing '11'
```
If anyone want to test it out, I prepared easy plug&play sandbox. It requires VS Code and Docker installed to launch dev container based of ros noetic docker image, where you can reproduce this error. More info in attached `readme.md`
[message_store_noetic.tar.gz](https://github.com/strands-project/mongodb_store/files/8832212/message_store_noetic.tar.gz)

ggorjup, 2022-10-03:
The package has now been updated and released for ROS Noetic - `mongodb_store` can be installed with:  
`sudo apt install ros-noetic-mongodb-store`

Closing issue.

closed 2022-10-03 by ggorjup