strands-project/mongodb_store #273: Change of maintainer

hawesie, 2022-08-04:
It may have become obvious, but I no longer have the capacity to check PRs or manage releases. Is anyone else willing to take this on, e.g. @marc-hanheide @knorth55 @furushchev @ggorjup @708yamaguchi ?

marc-hanheide, 2022-08-04:
not so keen myself. My quota of (poorly) maintained packages is beyond capacity already 😔

knorth55, 2022-08-05:
I can help checking PRs. But when I'm busy, I will be late to check it.

ggorjup, 2022-08-23:
I apologize for the late reply, I was on leave.

I can help with maintaining the repository with support of my company - FLX Robotics, since it is in our interest to maintain its quality.

hawesie, 2022-08-31:
@ggorjup that's great news, thanks! I think a key short-term task is to create a new release. Have you released ROS packages before?

ggorjup, 2022-09-05:
@hawesie that makes sense yes. I haven't released ROS packages before, but I went through the release instructions.

Did you have in mind to just create the Noetic release or also update Melodic?

In my PR I didn't touch `mongodb_log` and I see there are some Python 2 statements in the code - I guess this should be fixed before releasing right?

hawesie, 2022-09-06:
I think it's definitely time to create a Noetic release. 

And if you're able to fix the Python 2 statements in `mongodb_log` then great. Alternatively you could try releasing it and wait for someone else who needs to fix it.

ggorjup, 2022-09-07:
Sure, I'll fix the statements and make a PR before releasing.

ggorjup, 2022-09-07:
I made a PR updating the `mongodb_log` statements for Python 3 here: #277 
After it's merged, I'll start with the release process.

I didn't touch anything else, but I'd like to clean up the packages a bit after the release. I'll make a separate issue for that.

ggorjup, 2022-09-15:
I ran into an issue while running pre-release tests for Melodic - I addressed it here: #279 
I'd like to resolve this before releasing, so we don't run into any build/test issues in the process.

After merging, I'll open an initial PR to `rosdistro/noetic/distribution.yaml`, for the purpose of pre-release tests on Noetic.  
When I confirm that they pass, I'll continue with the standard `bloom` process.

Do I need maintainer rights for that or should I run `catkin_generate_changelog` & `catkin_prepare_release` on my fork and then open a PR here?

hawesie, 2022-09-15:
@ggorjup this is excellent, thanks! I've just made you an admin for the repo, so hopefully that's enough

ggorjup, 2022-09-16:
@hawesie thank you very much! I think that will be enough.  
I'll start with the release as soon as I confirm that the pre-release tests pass.

ggorjup, 2022-09-20:
@hawesie it seems I will also need write privileges for the release repository, if we wish to keep the same one: https://github.com/strands-project-releases/mongodb_store

marc-hanheide, 2022-09-20:
@ggorjup I invited you.

ggorjup, 2022-09-20:
Wow, that was fast! Thank you @marc-hanheide!

ggorjup, 2022-09-20:
The Noetic release PR is here: https://github.com/ros/rosdistro/pull/34588

The only issue I had is that the `buster` distribution has no `rosdep` key for `mongodb` (or `apt` key, for that matter) so I just skipped it when running `bloom-release`.

Is this something we can/should address?