strands-project/mongodb_store #280: Boost conflict on the noetic branch leads to segfaults

jagomo, 2022-10-28:
I've been toying a bit with the new noetic branch trying to update some old code base. In the same node where I make use of MessageStoreProxy I need to make use of boost::thread.

I was getting a segfault randomly happening in the spawned thread. After debugging for a while without a clue. Found out that the resulting node is linking to boost 1.71 (noetic dependency) and 1.67 (libmongoclient dependency). This is known to lead to segfaults in some cases.

https://answers.ros.org/question/190902/two-versions-of-libboost/

Even on the provided message_store_cpp_test this happens:

```
ldd devel/lib/mongodb_store/message_store_cpp_test | grep boost
	libboost_thread.so.1.67.0 => /lib/x86_64-linux-gnu/libboost_thread.so.1.67.0 (0x00007f70eb667000)
	libboost_system.so.1.67.0 => /lib/x86_64-linux-gnu/libboost_system.so.1.67.0 (0x00007f70eb660000)
	libboost_thread.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0 (0x00007f70eb5d9000)
	libboost_chrono.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_chrono.so.1.71.0 (0x00007f70eb5cb000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f70eb5ad000)
	libboost_regex.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_regex.so.1.71.0 (0x00007f70eb159000)
```

Is there any way to prevent this and force the usage of one boost library only?

jagomo, 2022-10-28:
Just to extend, the dependency conflict can be also seen here

´´´
❯ apt depends libmongoclient0
libmongoclient0
  **Depends: libboost-system1.67.0
  Depends: libboost-thread1.67.0**
  Depends: libc6 (>= 2.14)
  Depends: libgcc1 (>= 1:3.0)
    libgcc-s1
  Depends: libsasl2-2
  Depends: libssl1.1 (>= 1.1.0)
  Depends: libstdc++6 (>= 5.2)

❯ apt depends libmongoclient-dev 
libmongoclient-dev
  Depends: libmongoclient0 (= 1.1.3-3)
  Breaks: mongodb-dev (<< 2:1.1.2-1~)
  Replaces: mongodb-dev (<< 2:1.1.2-1~)

❯ apt depends ros-noetic-mongodb-store
ros-noetic-mongodb-store
  **Depends: libboost-thread1.71.0**
  Depends: libc6 (>= 2.14)
  Depends: libgcc-s1 (>= 3.0)
  Depends: libmongoclient0 (>= 1.1.3)
  Depends: libstdc++6 (>= 9)
  Depends: libmongoclient-dev
  Depends: mongodb
  Depends: python3-future
  Depends: python3-pymongo
  Depends: ros-noetic-geometry-msgs
  Depends: ros-noetic-mongodb-store-msgs
  Depends: ros-noetic-roscpp
  Depends: ros-noetic-rospy
  Depends: ros-noetic-std-msgs
  Depends: ros-noetic-std-srvs
  Depends: ros-noetic-topic-tools

´´´

jagomo, 2022-10-31:
After some more digging, realized that the C++ support is based on the legacy driver that went EOL some years ago. Are there any plans to support the current C++ driver? (it's boost-free so it would solve the problem), but the new driver has a completely different API:

https://stackoverflow.com/questions/45319979/upgrading-mongodb-c-driver-to-mongocxx-3-1-2

ggorjup, 2022-11-22:
Hi @jagomo, thanks for finding and investigating this issue.

Unfortunately there aren't any plans to support the new C++ driver at this time. The switch would also complicate dependency handling, since it seems the new C++ driver is not on `apt` like the old one (https://mongocxx.org/mongocxx-v3/installation/linux/).

That being said, if you find a solution for this issue, any PRs will be greatly appreciated.